serifpersia/esp32partitiontool #1: csv import not working on linux

tobozo, 2024-02-28:
hi and thanks for that great plugin :+1: 

here's a couple of csv files I tried to import and produced java exceptions:

```csv
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x480000,
app1,     app,  ota_1,   0x490000,0x480000,
spiffs,   data, spiffs,  0x910000,0x6E0000,
coredump, data, coredump,0xFF0000,0x10000,
```

```csv
# 6 Apps + Factory
# Name,   Type, SubType,    Offset,     Size
nvs,      data, nvs,        0x9000,   0x5000
otadata,  data, ota,        0xe000,   0x2000
ota_0,    0,    ota_0,     0x10000, 0x200000
ota_1,    0,    ota_1,    0x210000, 0x200000
ota_2,    0,    ota_2,    0x410000, 0x200000
ota_3,    0,    ota_3,    0x610000, 0x200000
ota_4,    0,    ota_4,    0x810000, 0x200000
ota_5,    0,    ota_5,    0xA10000, 0x200000
firmware, app,  factory,  0xC10000, 0x0F0000
spiffs,   data, spiffs,   0xD00000, 0x2F0000
coredump, data, coredump, 0xFF0000,  0x10000
```

```log
Exception in thread "AWT-EventQueue-0" java.lang.NumberFormatException: For input string: "0x2000"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Integer.parseInt(Integer.java:580)
	at com.serifpersia.esp32partitiontool.FileManager.hexToKB(FileManager.java:153)
	at com.serifpersia.esp32partitiontool.FileManager.updateUIComponents(FileManager.java:147)
	at com.serifpersia.esp32partitiontool.FileManager.importCSV(FileManager.java:113)
```

I can build this plugin successfully (just tested the build.sh) so feel free to ask if you need testing.

tobozo, 2024-02-28:
Possible reasons why this errors:

- there can be more than one line of comments in the csv file
- line may or may not be terminated with a comma
- column values can contain spaces and should be trimmed

serifpersia, 2024-02-28:
> hi and thanks for that great plugin 👍
> 
> here's a couple of csv files I tried to import and produced java exceptions:
> 
> ```
> # Name,   Type, SubType, Offset,  Size, Flags
> nvs,      data, nvs,     0x9000,  0x5000,
> otadata,  data, ota,     0xe000,  0x2000,
> app0,     app,  ota_0,   0x10000, 0x480000,
> app1,     app,  ota_1,   0x490000,0x480000,
> spiffs,   data, spiffs,  0x910000,0x6E0000,
> coredump, data, coredump,0xFF0000,0x10000,
> ```
> 
> ```
> # 6 Apps + Factory
> # Name,   Type, SubType,    Offset,     Size
> nvs,      data, nvs,        0x9000,   0x5000
> otadata,  data, ota,        0xe000,   0x2000
> ota_0,    0,    ota_0,     0x10000, 0x200000
> ota_1,    0,    ota_1,    0x210000, 0x200000
> ota_2,    0,    ota_2,    0x410000, 0x200000
> ota_3,    0,    ota_3,    0x610000, 0x200000
> ota_4,    0,    ota_4,    0x810000, 0x200000
> ota_5,    0,    ota_5,    0xA10000, 0x200000
> firmware, app,  factory,  0xC10000, 0x0F0000
> spiffs,   data, spiffs,   0xD00000, 0x2F0000
> coredump, data, coredump, 0xFF0000,  0x10000
> ```
> 
> ```
> Exception in thread "AWT-EventQueue-0" java.lang.NumberFormatException: For input string: "0x2000"
> 	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
> 	at java.lang.Integer.parseInt(Integer.java:580)
> 	at com.serifpersia.esp32partitiontool.FileManager.hexToKB(FileManager.java:153)
> 	at com.serifpersia.esp32partitiontool.FileManager.updateUIComponents(FileManager.java:147)
> 	at com.serifpersia.esp32partitiontool.FileManager.importCSV(FileManager.java:113)
> ```
> 
> I can build this plugin successfully (just tested the build.sh) so feel free to ask if you need testing.

I tested on linux and its probably same on windows I found small label not updating correctly I have fix for that, the issue you are having is you have 11 items/rows. The tool supports only 10 at the moment. I would have to implement user generated based rows so you could create however many rows of partitions but for now its only 10 and I assume 11 bigger than 10 is cause of the issue here.

tobozo, 2024-02-28:
duly noted for the file with 11 entries, however the issue also happens with smaller csv files picked from the espressif tools folder such as this one:

```csv
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x480000,
app1,     app,  ota_1,   0x490000,0x480000,
spiffs,   data, spiffs,  0x910000,0x6E0000,
coredump, data, coredump,0xFF0000,0x10000,
```

serifpersia, 2024-02-28:
Yes the issue there is incompatible csv format spacing needs to be consistent, and I don't think last partition having missing comma would cause any issue. Try to correct the spacing. Maybe create a csv file from the tool and compare the differences.

tobozo, 2024-02-28:
I tried to correct the code and got some results with a smaller csv file of 9 lines (plus 2 lines of comments):

[image]

here's a diff, I can make a PR if you want:

```diff
diff --git a/src/main/java/com/serifpersia/esp32partitiontool/FileManager.java b/src/main/java/com/serifpersia/esp32partitiontool/FileManager.java
index 2d2b7b2..e0e8250 100644
--- a/src/main/java/com/serifpersia/esp32partitiontool/FileManager.java
+++ b/src/main/java/com/serifpersia/esp32partitiontool/FileManager.java
@@ -107,6 +107,7 @@ public class FileManager {
                                int rowIndex = 0;
                                String line;
                                while ((line = reader.readLine()) != null && rowIndex < ui.getNumOfItems()) {
+                                 if( line.startsWith("#") ) continue;
                                        String[] columns = line.split(", ");
                                        if (columns.length >= 5) {
                                                setUIComponents(rowIndex, true);
@@ -139,11 +140,11 @@ public class FileManager {
                JTextField partitionSizeHex = ui.getPartitionSizeHex(rowIndex);
                JTextField partitionOffset = ui.getPartitionOffsets(rowIndex);
 
-               partitionName.setText(columns[0]);
-               partitionType.setSelectedItem(columns[1]);
-               partitionSubType.setText(columns[2]);
-               partitionOffset.setText(columns[3].substring(2));
-               String hexSize = columns[4].substring(2);
+               partitionName.setText(columns[0].trim());
+               partitionType.setSelectedItem(columns[1].trim());
+               partitionSubType.setText(columns[2].trim());
+               partitionOffset.setText(columns[3].trim().substring(2));
+               String hexSize = columns[4].trim().substring(2);
                String kbSize = hexToKB(hexSize);
                partitionSize.setText(kbSize);
                partitionSizeHex.setText(hexSize);
@@ -1020,4 +1021,4 @@ public class FileManager {

```

serifpersia, 2024-02-28:
@tobozo Try latest repo change, the csv file that has <=10 partitions entries should be correctly imported. There is some ui touch that needs to be done but the import should work ui I will redo later as for more 10 entries thats not accounted scenario and it would have to redone from ground up since this java swing ui is static.

tobozo, 2024-02-28:
:warning:  the header can have multiple lines of comments or none at all

```
Invalid CSV format: Insufficient columns in the header.
```

serifpersia, 2024-02-28:
> ⚠️ the header can have multiple lines of comments or none at all
> 
> ```
> Invalid CSV format: Insufficient columns in the header.
> ```
> 
> [edit] and there can be comments in the middle of the csv flow too, comment contents should not be trusted

Pushed latest change should handle comments as well. Let me know if there is something else wrong with csv

tobozo, 2024-02-28:
latest change fixed the error but now the first row in my csv is missing.

[image]

I also observed that the Flash Size isn't updated after loading a csv.
Loading a csv based on 16MB flash should adjust the selected value in the select menu to 16MB too.

[image]

tobozo, 2024-02-28:
just use `if( line.startsWith("#") ) continue;` inside the while() loop in processCSV()

importCSV() should not consume any csv line and let processCSV() do the processing/filtering work

serifpersia, 2024-02-28:
I will look into the skip thing. Yes the flash size doesn't update but I think I can maybe calculate and update based on last hex offset

tobozo, 2024-02-28:
> maybe calculate and update based on last hex offset

that value is already available in the last cell

[image]

also `32MB` is missing from the list

[image]

serifpersia, 2024-02-28:
> that value is already available in the last cell [...]

I added support for 32MB, flash size is updating to correct value based on last offset(converted to decimal then / 1024 /1024 to get MB value thanks for noticing that I wouldn't.

Feel free to create pull request for the comment filtering I don't have time or motivation to do that since its was never intended to import not compatible formatted csv files(the idea was to create the csv file in the tool and use that one for import)

serifpersia, 2024-02-29:
I pushed latest change, as long as you have correctly formatted csv file (spacing & comments don't matter) ie your types are not numbers but text data or app the import will work without issue. I added 15 partitions support but I will look into making it reactive(+- buttons to add or remove partitions instead of static 15 total). The UI is clearing and updating correctly based on csv data and last selected partition hex offset. Feel free to reopen this if you find some other issue with CSV import, as well as open new issue regarding other features of the tool that have issues. 
Thanks again!

closed 2024-02-29 by serifpersia

tobozo, 2024-02-29:
thanks !

I'll try to fix the remaining bug on my fork (still crashing when CSV has no comments at all).

Does the `Bad CSV format` in the readme use integer values instead of strings in the `partition type` column?

According to [gen_esp32part.py](https://github.com/espressif/arduino-esp32/blob/master/tools/gen_esp32part.py#L37) this is a valid CSV and could be handled, so I'll also try to correct this.

expect a PR soon :-)

serifpersia, 2024-02-29:
> thanks !
>
> I'll try to fix the remaining bug on my fork (still crashing when CSV has no comments at all).
>
> Does the `Bad CSV format` in the readme use integer values instead of strings in the [...]

Bad CSV is when you have type as numbers and not data or app for type of partition, unless you think thats standard practice as well I could make it work with numbers instead of text app or data for partition type. I haven't tested no comments and no header,edit just did seems like it skips first partition entry but doesn't crash. Do most csv files have header file? Anyways feel free to do the PR if you are not satisfied with import csv support. Main issue with spacing was fixed.

tobozo, 2024-02-29:
please don't add more commits as I'll submit a PR soon

according to the contents of the csv files bundled with espressif core, the `type` column can have either of those values: `app`, `data`, `0`, `1`, `0x0`, `0x1`, `0x00`, `0x01`.

moreover, some partition names are hard-linked to some partition types (e.g. nvs and spiffs are always `data`).

while it's probably not the applet's job to maintain consistency between columns values (gen_partitions.py does some additional checks on the csv contents), it's still possible to extend the logic to handle string/hex/dev values; I'll include this with my PR along with other changes and bugfixes.

tobozo, 2024-02-29:
whoops apparently `createSPIFFS()` doesn't handle LittleFS and FFat partitions.

while mkfatfs requires manual installation, mklittlefs is bundled with espressif tools; so the next task after submitting the PR will be to research how this can be handled.

serifpersia, 2024-02-29:
> whoops apparently `createSPIFFS()` doesn't handle LittleFS and FFat partitions.
> 
> while mkfatfs requires manual installation, mklittlefs is bundled with espressif tools; so the next task after submitting the PR will be to research how this can be handled.

I'm not sure whats the difference with different partition styles, I used the code from esp32fs  plugin for spiffs creation. What are the benefits of using non spiffs?

tobozo, 2024-02-29:
spiffs is bugged, slow and inefficient, using it isn't recommended as it creates a bigger risk of data corruption

LittleFS solves many of spiffs problems and delivers better performances.

tobozo, 2024-02-29:
idea based on `less is more` postulate: 

- completely remove "Flash SPIFFS" feature from the applet, it is redundant with ESP32 Sketch Data Upload utility, which already covers all know filesystem types
- remove "flash sketch", or rename it to something like "write and close" so the flashing can be done from the Arduino IDE using the default "partition.csv" file from the sketch folder
- remove "flash merged bin" since it is redundant with Arduino IDE features once a partitions.csv file is created in the sketch folder

tobozo, 2024-02-29:
I haven't commited yet, but this is where I am after applying a few changes, which I'm ready to discuss:

- added bottom and right margins
- assigned colors by partition type+subtype
- removed spiffs tools from the view
- made the window not resizable

[image]

serifpersia, 2024-02-29:
> I haven't commited yet, but this is where I am after applying a few changes, which I'm ready to discuss:
>
>     * added bottom and right margins
>
>     * assigned colors by partition type+subtype
>
>     * [...]

Looks good. I would still like to keep the Flash SPIFFS somewhere in the plugin, I would like the tool to have these tools built in, the idea of relying on separate tool for spiffs is not working for me. The SPIFFS code is very similar from that tool anyways.  The idea of this tool is to be ultimate tool for doing custom partitions for your board as well as quick upload to test it on your esp32 board.

tobozo, 2024-02-29:
I understand, now how would you implement LittleFS in the current view? 
mklittlefs and mkspiffs use the same syntax, so it's just a matter of matching the corresponding executable file; what about adding a combo box with selectable choice SPIFFS/LittleFS?

[image]

serifpersia, 2024-02-29:
Yes a combo box to select type of storage default littlefs since you said its better but for legacy spiffs can also be selected. The Flash sketch can be removed completely as you said. Merge bin button can just create merge binary(useful for project release) I was thinking to have a single loop of logic for merge bin just prompt users to type the offset for bootloader since its different for each board for s2 its 0x1000 for s3 0x0. Maybe we should discuss this changes in your PR. I can start doing my merge bin changes when you commit your finalized PR.

tobozo, 2024-02-29:
bootloader offset can be retrieved with `BaseNoGui.getBoardPreferences().get("build.bootloader_addr")` so I'll implement that, sounds safer than having an editable field

I'll remove the whole Flash Sketch mvc and implement the spiffs/littefs combo