serifpersia/esp32partitiontool #4: feature request: PlatformIO

jankoscharf, 2024-03-09:
Hi,
nowerdays PlatformIO is my tool of choice over the Arduino IDE.
Will it be possible to use [esp32partitiontool](https://github.com/serifpersia/esp32partitiontool) with PlatformIO in the future?

Thanks.

serifpersia, 2024-03-09:
@jankoscharf 
Hi there.
There is possibility for it to work externally or be integrated as tool in PlaltformIO environment, the execution and file management would have to be rewritten in that case. 
Please suggest how will that work since I'm not familiar with it. How does it work where is the esp32 sdk located and so on.
Making it work externally as jar application would be easiest without integration to any IDE or platform, you would just provide the esp32 sdk tools folder.  As I said  if you have info you could share that could help feel free to do so, but without any information on how PlatformioIO does things with esp32 I'm afraid I can't start working on this idea.

jankoscharf, 2024-03-09:
I have never worked with java nor am I familiar with vscode extensions (which PlatformIO also is one). Maybe this helps a little: https://code.visualstudio.com/docs/java/extensions
I think the "ESP-IDF" extension installation process defines the location where the esp32 sdk is located.

serifpersia, 2024-03-09:
What framework are you using in PlatformIO? I'm trying it out right now with Arduino framework to see how it does stuff.

jankoscharf, 2024-03-09:
Yes, Arduino framework. I like to use the Arduino third party libs.

serifpersia, 2024-03-09:
Sure I will look into this maybe I can make it work with Arduino framework inside PlatformIO maybe as external jar I don't think I can integrate as tool as I did in Arduino IDE unless you have resources for that.

jankoscharf, 2024-03-09:
Sadly no. I would have to start at zero. IMHO a standalone app would be an option I think. As far as I know java is perfectly flexible to run on all the important platforms. I use Linux only...

serifpersia, 2024-03-09:
Yes even this tool runs on any platform haven't tried mac bc I don't have it. But yes I think external jar you can execute would work. All you need to do is provide  esp32 sdk folder so it can use tools for creating partitions.bin file spiffs and merge commands. You would have to have python3 as well. First step would be to get UI to work as external app then add functionally back to the UI.

jankoscharf, 2024-03-09:
Platformio will use a *.csv partition file located in a project source tree which is configured in the platformio.ini:
```
[env:esp32]
platform = espressif32
board = <YOUR ESP32 BOARD>
framework = arduino

board_build.partitions  = MY_PARTITION_FILE.csv
```
So one can keep the partition file to use in the source tree, no need to install in the espressif path explicitly.

serifpersia, 2024-03-09:
> Platformio will use a *.csv partition file located in a project source tree which is configured in the platformio.ini:
>
> ```
> [env:esp32]
> platform = espressif32
> board = <YOUR ESP32 BOARD>
> framework = [...]

I'm talking about esp32 sdk tools directory. We need tools (esptool partition gen tool spiffs tool etc) to create output files. But thanks for the tip about csv location. So far I got UI to be standalone and im trying out python command to see if it can execute without any IDE There will be a console output window so you can see whats going on
[image]

tobozo, 2024-03-09:
> So far I got UI to be standalone

nice! I tried to do that without success, how did you manage?

serifpersia, 2024-03-09:
> > So far I got UI to be standalone
> 
> nice! I tried to do that without success, how did you manage?

I'm doing it from ground up. The main issues were loading of csv file & updating the ui with that data and resources images. I added the resources images to src dir as well as making it load default csv content for now. Load and save thing i removed(I removed all code from fileManager I'm adding stuff back as I go)
[image]

I was able to get python to work on windows at least next is log output and will try to create partitions.bin for test.
[image]
As for python we will use the one provided with platformio most users have global python installed but I won't ask users to have it so we will use already included one in this dir for windows %HOMEPATH%\.platformio\python3 linux ~/.platformio/python3
[image]

tobozo, 2024-03-09:
Configuration complexity is increased with platformio.ini, the applet will need to work with a config extractor and get the paths from an argument list.

The best pick in order to have platformio to collaborate with java (e.g. provide additional paths to src, sdk and build folders) is to use the [platformio config api](https://github.com/platformio/platformio-core/blob/develop/platformio/project/config.py) from a python script that is referenced in the ini file.

The python script runs after library resolving and before compiling the sketch, and makes it possible to insert a modal call the java applet with argument values collected from python.

##### platformio.ini
```ini
[env:ESP32PartitionTool]
extra_scripts = ESP32PartitionTool.py
```

##### ESP32PartitionTool.py
```python
Import('env')
from os.path import join, realpath
import subprocess

board = env.BoardConfig()
build_core       = board.get("build.core", "").lower()
partition        = board.get("build.partitions")
build_variant    = board.get("build.variant")

print( "****************************************************" )
print( "Build Core    = " + build_core )
print( "Partition     = " + partition )
print( "Build Variant = " + build_variant )
print( "****************************************************" )

command = 'java -cp "/path1:/path2" -jar "/path/to/ESP32PartitionTool.jar" "' + partition + '"'
return_code = subprocess.call(command, shell=True)

```

serifpersia, 2024-03-09:
> Configuration complexity is increased with platformio.ini, the applet will need to work with a config extractor and get the paths from an argument list.
> 
> The best pick in order to have platformio to collaborate with java (e.g. provide additional paths to src, sdk and build folders) is to use the [platformio config api](https://github.com/platformio/platformio-core/blob/develop/platformio/project/config.py) from a python script that is referenced in the ini file.
> 
> The python script runs after library resolving and before compiling the sketch, and makes it possible to insert a modal call the java applet with argument values collected from python.
> ##### platformio.ini
> 
> ```ini
> [env:ESP32PartitionTool]
> extra_scripts = ESP32PartitionTool.py
> ```
> 
> ##### ESP32PartitionTool.py
> 
> ```python
> Import('env')
> import subprocess
> 
> class_path       = "/path1:/path2"
> jar_path         = "/path/to/ESP32PartitionTool.jar"
> board            = env.BoardConfig()
> build_core       = board.get("build.core", "").lower()
> partition        = board.get("build.partitions")
> build_variant    = board.get("build.variant")
> 
> print( "****************************************************" )
> print( "Build Core    = " + build_core )
> print( "Partition     = " + partition )
> print( "Build Variant = " + build_variant )
> print( "****************************************************" )
> 
> 
> command_format = 'java -cp "%s" -jar "%s" "%s" "%s" "%s"'
> command = command_format % (class_path, jar_path, partition, build_variant, build_core)
> 
> print (command)
> 
> return_code = subprocess.call(command, shell=True)
> ```

I'm going to create a initial commit in the new branch , if you want to contribute to this as well you can do a pr in that branch. I will set some basics to get started but later on board type and bootloader offests for sppifs and merge upload commands would need to be implemented.

tobozo, 2024-03-09:
> I will set some basics to get started but later on board type and bootloader offests for sppifs and merge upload commands would need to be implemented.

SPIFFS/createBin/mergeBin buttons won't be needed since the applet will be called in the middle of a platformio build flow where the next step is to create said binaries.

Those buttons only make sense with Arduino IDE, because the Sketch Data Upload in not a standard Arduino IDE tool and requires another java plugin, but platformio already handles the data upload with the `uploafs` target flag.

serifpersia, 2024-03-09:
Oh I see well. I will commit what I got now. Some basics done for starters, both linux and windows  scripts produce runnable jar. We will go from there I guess.

tobozo, 2024-03-09:
> both linux and windows scripts produce runnable jar

can you either confirm `java -jar /path/to/ESP32PartitionTool.jar` is enough to run the app or provide an example of the necessary command line?

I can see the new `standalone` branch but I don't see any changes for the standalone version there, so there's not much I can do.

serifpersia, 2024-03-09:
> can you either confirm `java -jar /path/to/ESP32PartitionTool.jar` is enough to run the app or provide an example of the necessary command line?
>
> I can see the new `standalone` branch but I don't [...]

I deleted that branch and created new one with better name. The changes I've made are live. Yes after building it with build.sh you can run it in output folder java -jar jar name.jar

tobozo, 2024-03-09:
thanks, I already got some results by turning the repository into a platformio package

```json
{
  "name": "tool-esp32partitiontool",
  "version": "1.4.0",
  "description": "Partition utility gui (requires java) to create custom partition table",
  "keywords": ["tools","build tools","filesystem"],
  "license": "MIT",
  "system": ["linux_x86_64"],
  "repository": {"type": "git","url": "https://github.com/serifpersia/esp32partitiontool"}
}
```

However platformio always uses the main branch or the latest release and this will eventually collide with Arduino-IDE context, so I wonder how those build variants will be maintained in the future.

If both versions could coexist in the same codebase it would prevent dissociation and reduce the workload.

Real world example: if the next natural thing to do with a popular app is to translate the GUI (implementing l10n), you don't want to integrate twice every translatable label.

serifpersia, 2024-03-09:
Should the platformio project move to new repo then?

tobozo, 2024-03-09:
> Should the platformio project move to new repo then?

I don't like this idea, it means two projects to maintain, the tech debt is depressing. 

I'm researching a way to have both versions to coexist, added two java files `InputArduino.java` and `InputPlatformio.java` with a duplicate class name:

```java
package com.serifpersia.esp32partitiontool;

final class AppSettings {
  public AppSettings() {
  }
}
```

then I can use one of these shell scripts to build for platformio or arduino without getting the duplicate class error:

#### compile jar for platformio
```shell
# Find Java files and exclude InputArduino.java, write to 'class' file
find src/main/java/com/serifpersia/esp32partitiontool -type f \( ! -iname "InputArduino.java" \) > class
# Call the compiler and specify the 'class' file instead of source folder
javac -Xlint -d "$bin_folder" @class
```

#### compile jar for arduino
```shell
# Find Java files and exclude InputPlatformio.java, write to 'class' file
find src/main/java/com/serifpersia/esp32partitiontool -type f \( ! -iname "InputPlatformio.java" \) > class
# Call the compiler and specify the 'class' file instead of source folder
javac -Xlint -d "$bin_folder" @class
```

it's probably possible with windows too, found an example on SO but can't test it:

```bat
@echo off
for %%i in (src/main/java/com/serifpersia/esp32partitiontool/*.java) do if /i not "%%~i" == "InputArduino.java" javac "%%~i"
```

tobozo, 2024-03-10:
finally found a project layout that allows me to create both Tool-implementation and standalone jar files.

[image]

I still don't have a clear idea of what was removed from the arduino version, so I'll just try to restore it to what is strictly necessary for both scenarios, which means I'll probably get rid of the redundant SPIFFS/mergeBin/createBin features that platformio/arduino tools already handle.

I observed are a few minor problems with the standalone version rendering though: some text elements (buttons, labels, combos) are using bold font, text in the setup window becomes truncated.

[image]

And the help window has a duplicate decoration :rofl: 

[image]

tobozo, 2024-03-10:
@jankoscharf before going further I'd be happy to have your feedback on that **temporary(*)** implementation, it requires to create a new python file in the project folder and add a new environment in platformio.ini; obviously it also requires java.

(*) The url to the `platform_packages` zip file is pointing to a test release on my fork, and that fork will eventually be deleted after the code is merged.

#### platformio.ini
```ini
[env:build-partition]
platform               = espressif32
board                  = esp32dev
extra_scripts          = partition_manager.py
platform_packages      = tool-esp32partitiontool @ https://github.com/tobozo/esp32partitiontool/releases/download/v1.4.1/esp32partitiontool-platformio.zip

```

#### partition_manager.py
```python
Import('env')
import os.path
import sys

# add esp32partitiontool to path
sys.path.append(os.path.abspath( env.PioPlatform().get_package_dir("tool-esp32partitiontool") ))
# import module
from esp32partitiontool import *
# run module
load_pm(env)
```

jankoscharf, 2024-03-10:
I implemented it on my machine and it works! Great to see!
I would like to recommend https://docs.platformio.org/en/latest/scripting/custom_targets.html#command-shortcut
This will give you a "Custom" menu entry in your "Project Tasks" where you can run the script.
The way it is integrated at the moment will be annoying very fast because you have to close the esp32partitiontool window every time you compile/upload.

tobozo, 2024-03-10:
@jankoscharf awesome tip, thanks! 

I've updated the test package as per your suggestion, the custom target name is `edit_partition`

    pio run -e build-partition -t edit_partition

[image]

serifpersia, 2024-03-11:
I'm not sure if I'm doing something wrong or the script .py is not working well on windows but whenever I select the build-partition env i get errors and i cant seem to run the edit_partition task
[image]
[image]

I'm using your hello world example with changes to board and memory size and pointing to default.csv file
Here is my log:

```ini
[platformio]
src_dir                 = src
default_envs            = hello-world

[env]
framework               = arduino
platform                = espressif32
board                   = esp32s2dev
board_build.partitions  = partitions/default.csv
board_upload.flash_size = 4MB

[env:hello-world]
upload_speed            = 1500000
monitor_speed           = 115200

[env:build-partition]
extra_scripts          = partition_manager.py
; platform_packages      = tool-esp32partitiontool
platform_packages      = tool-esp32partitiontool @ https://github.com/tobozo/esp32partitiontool/releases/download/v1.4.1/esp32partitiontool-platformio.zip
```

tobozo, 2024-03-11:
task `edit_partition` not being available in vscode is a consequence of the python script error

looks like `env.PioPlatform().get_package_dir("tool-esp32partitiontool")` returned `None`, which is strange since the tool appears to be installed twice:

    Tool Manager: tool-esp32partitiontool@1.4.0 has been installed!

I've created a new test release with incremented version number (v1.4.2) and updated example, I hope this one will register better in vscode 🤞

The python script has been updated in the example, and the URL to the plugin has changed accordingly in platformio.ini: 

https://github.com/tobozo/esp32partitiontool/releases/download/v1.4.2/esp32partitiontool-platformio.zip

tobozo, 2024-03-11:
I've tested an even simpler ini file, package and target registration are hoisted to the top environment, so that the `edit_partition` target can be used against any environment:

```ini
[platformio]
src_dir                 = src
default_envs            = hello-world

[env]
framework               = arduino
platform                = espressif32
board                   = esp32dev
; register the "tool-esp32partitiontool" package
platform_packages       = tool-esp32partitiontool @ https://github.com/tobozo/esp32partitiontool/releases/download/v1.4.2/esp32partitiontool-platformio.zip
; register the "edit_partition" target
extra_scripts           = partition_manager.py

[env:hello-world]
board_build.partitions  = partitions/defaultcsv
board_upload.flash_size = 4MB
upload_speed            = 1500000
monitor_speed           = 115200
```

serifpersia, 2024-03-11:
I managed to get it working only with the manual method I had to modifiy py script for my platform windows instead of linux and i was able to use the esp32partition task

```ini
[platformio]
src_dir                 = src
default_envs            = hello-world

[env]
framework = arduino
platform = espressif32
board = esp32dev
; register the "tool-esp32partitiontool" package
;platform_packages       = tool-esp32partitiontool @ https://github.com/tobozo/esp32partitiontool/releases/download/v1.4.2/esp32partitiontool-platformio.zip
; register the "edit_partition" target
platform_packages      = tool-esp32partitiontool
extra_scripts           = partition_manager.py

[env:hello-world]
board_build.partitions  = partitions/default.csv
board_upload.flash_size = 4MB
upload_speed            = 1500000
monitor_speed           = 115200
```
```json
{
  "name": "tool-esp32partitiontool",
  "version": "1.4.2",
  "description": "Partition utility gui (requires java) to create custom partition table",
  "keywords": ["tools","build tools","filesystem"],
  "license": "MIT",
  "system": ["windows_amd64"],
  "repository": {"type": "git","url": "https://github.com/serifpersia/esp32partitiontool"}
}
```

serifpersia, 2024-03-11:
@tobozo  I managed to make it work the only change was to version release and removal of system parameter in json I made a test release and directed it to that url and it worked
[image]

tobozo, 2024-03-11:
I have modifyied create_platformio_package.sh to overwrite the `version` value in `package.json` according to the tag name before creating the archive, so keep in mind the name of the tag you'll create from the release page before publishing a new release is very important, it must be semver compliant and have a version number superior to the last version (e.g current version is [1.4.4](https://github.com/tobozo/esp32partitiontool/releases/tag/v1.4.4) and next version should be 1.4.5)

serifpersia, 2024-03-12:
@tobozo  Could you look into why does it fail at this task. I'm not able to create auto release. Both package.json version 1.4.5 matches with the tag I give it but it fails. For some reason it deleted the 1.3 release as well...
Run softprops/action-gh-release@v2
  with:
    files: /home/runner/work/esp32partitiontool/esp32partitiontool/archives/**
  
    token: ***
⚠️ Unexpected error fetching GitHub release for tag refs/tags/v1.4.5: HttpError: Resource not accessible by integration
Error: Resource not accessible by integration

tobozo, 2024-03-12:
1.4 looks in a good shape :+1: 

what's for the next iteration? 

- fix platformio design (help window still strange and bold coming out of nowhere)
- automate `pio pkb publish` to get the package in platformio registry, also it simplifies the platformio.ini (no more URL needed, version selection possible)
- restore arduino-specific buttons/panels

serifpersia, 2024-03-12:
I'm closing this as complete, tested both tools in both platforms and no issues. I had issues with versioning it didn't like x.x hs to be x.x.x for it to work as well as arduino zip has  to be specific casings idk why that should matter. With x.x versioning it was using old build from 10/03  and not zip from the release but doing 1.4.0 fixed that. 
For Arduino specific things a new button will be next to help/about to open a dialog there the spiffs and merge tooling will reside. 
Once again nice job @tobozo  I wouldn't have done it as tool integration like you did instead standalone app you execute out side of your IDE of choice.

closed 2024-03-12 by serifpersia