serifpersia/esp32partitiontool #12: Unable to use as after the unpacking it doesn't appear in the tools drop down menu in the Arduino IDE.

The-Valveman, 2024-06-12:
I have downloaded the Arduino version from the assets folder in the releases menu then followed the instructions in the read me page but when I open the Arduino IDE and open the tools drop down this tool does not appear. when you say "unpack the tool to the tools directory" I presume you mean unzip the tool to the tools folder this I have done and the path looks like the one in the example.
I have restarted the Arduino IDE several times but it still doesn't appear 

**Expected behavior**
I expected the tool to appear in the tools drop down menu as per your screenshots.

**Desktop (please complete the following information):**
 I am using Windows 10 and Arduino IDE 1.8.19

**Additional context**
I am using an ESP32 with an 8MB flash and need more than 3MB for the app I am installing onto the ESP32 I don't need the OTA and only need a small SPIFFS partition. I am using the Arduino IDE as i have used this for a long time and at being nearly 60 i dont want to have to take on the steep learning curve of moving over to PlatformIO

tobozo, 2024-06-12:
hi,

missed a subfolder maybe?

your Arduino folder may be named differently on windows but this is how it's organized on a linux machine:

[image]

serifpersia, 2024-06-13:
@The-Valveman Your tools directory structure has to match the following:
 `Arduino home dir>tools>ESP32PartitionTool>tool>ESP32PartitionTool.jar`. If you don't have tools folder and only libraries in your Arduino folder create one and simply extract the release arduino zip there it should match the mentioned path. Your Arduino directory can be found in My Documents under Windows OS and on linux its user home directory .arduino. You might have to enable show hidden files to see it. On the OS X you can the tools directory in:
  `~/Documents/Arduino/ ` and unzip the release arduino zip there. That should cover it.

closed 2024-06-13 by serifpersia