serifpersia/esp32partitiontool #13: Simplified Chinese is not supported.

changxinchang, 2024-07-12:
Simplified Chinese is not supported.
[image]

serifpersia, 2024-07-12:
I verified the issue seems like other languages are also affected. This is due to using DejaVu font which should work fine I tested the same text using that font in Notepad and it displayed all text fine but it doesnt work in the tool. The immediate solution is to use built in fonts that display the text correctly instead. @tobozo If you have time investigate this font issue. The following fonts work with these languages.

```java
	public UI(JFrame frame, String title) {

		// Show tool tips immediately
		ToolTipManager.sharedInstance().setInitialDelay(0);

		defaultFont = new Font("Monospaced", Font.PLAIN, 12);
		monotypeFont = new Font("Monospaced", Font.PLAIN, 12);
		monotypeBoldFont = new Font("Monospaced", Font.BOLD, 12);
		condensedFont = new Font("Monospaced", Font.PLAIN, 12);

		setFrame(frame, title);
		setLayout(new BorderLayout(0, 0));
		setOpaque(false); // transparent background!
		init();
	}
```
[image]

changxinchang, 2024-07-13:
How to modify the downloaded ESP32PartitionTool.jar

serifpersia, 2024-07-13:
> How to modify the downloaded ESP32PartitionTool.jar

There is nothing you can edit/modify in the jar its release binary, I will use different font and release new version you can download that later. Could you look into the translation files for your language and see if its correct so I can update that as well. You can find the translation files here. [english](https://github.com/serifpersia/esp32partitiontool/blob/main/src/main/resources/l10n/labels.properties) and [chinese](https://github.com/serifpersia/esp32partitiontool/blob/main/src/main/resources/l10n/labels_zh.properties)
If you feel like the translation is wrong please fork and do a PR with the changes to the chinese file.

closed 2024-07-13 by serifpersia