serifpersia/esp32partitiontool #15: are you going to port it to Arduino 2.x ?

laverdanny, 2025-03-12:


serifpersia, 2025-03-13:
Sadly this isn't on my list of things I want to tackle. Arduino 2.x is built with electron and doesn't use Java which this project uses for UI and logic that calls Arduino IDE tools for upload.

 As this is high difficulty task I do not plan to do this rewrite.
You can still use it in legacy 1.8 or modern platformio(partition csv creation, that framework has built in tools for filesystem like littlefs etc). 

If you have any further questions you can reply, I will close this shortly after.

laverdanny, 2025-03-13:
hi
thanks for answering back
Actually I discovered (https://www.youtube.com/watch?v=EuHxodrye6E)) that starting with Arduino 2.x we can put a partitions.csv in sketch folder (well, it didn't work for me, but I have troubles with IDE plugins too, it's maybe my own problr, digging on it)

In such a case your soltion is still good, just have to put result in sketch folder too

Regards

closed 2025-03-13 by serifpersia