serifpersia/esp32partitiontool #16: Tool erase all ESP32 flash after usage. Desired option to not erase flash.

aldock, 2025-06-03:
**Describe whats wrong**
Want to keep flash content when using the tool, but it is erasing all content. I´m using Arduino IDE 1.8.19

**To Reproduce**
I have wrote a file at the flash and want to keep it.

serifpersia, 2025-06-04:
> **Describe whats wrong** Want to keep flash content when using the tool, but it is erasing all content. I´m using Arduino IDE 1.8.19
> 
> **To Reproduce** I have wrote a file at the flash and want to keep it.

I assume you are using the Merge binary and upload button. This creates merge bin file that gets flashed at 0x0 offset so by this very nature its erasing the whole flash. 

There is nothing that can be done for as this is how merge binary works when creating merge binary with firmware spiffs and other partitions that get included. If you need to upload only firmware or filesystem(spiffs) you could either use arduino ide's upload button(make sure to use Huge APP partition scheme to let arduino ide use custom partitions csv table you generated or you use merge bin and not the merge bin and upload button - this way you generate all partitions + merge binary and you can upload whatever partition you want via esp cmd line or some other esp32 flashing program. 

For spiffs( filesystem) you can use upload filesystem only button if you only need to update filesystem partition.  

Insight to what are you trying to do would help me to point you to the right direction. But no this is not an issue this is how merge binary and upload works it bundles all partitions for your specific board with specific offests and uploads it ad 0x0 which does erase everything.

aldock, 2025-06-04:
Thank you!  I´ll follow this way.

closed 2025-06-04 by serifpersia